dionysos-dev/Dionysos.jl #26: First-order approximation of reachable set

guberger, 2020-08-05:
FYI: I am starting with the implementation of a method to compute a "first-order" outer-approximation of the reachable set, that is using a polyhedron, and not only a HyperRectangle (with growth bound function...). In a first time, I am not sure to be able to account smartly for disturbance/noise, and thus I will neglect this effect.

blegat, 2020-12-08:
From a chat with @mforets: We may be able to leverage discretization work from ReachabilityAnalysis: https://github.com/JuliaReach/ReachabilityAnalysis.jl/blob/master/src/Continuous/discretization.jl

mforets, 2022-07-05:
> From a chat with @mforets: We may be able to leverage discretization work from ReachabilityAnalysis: https://github.com/JuliaReach/ReachabilityAnalysis.jl/blob/master/src/Continuous/discretization.jl

Just FYI we have since then written a review article about discretization methods, see https://arxiv.org/abs/2111.01454, and they are implemented in the library. If you find that useful, we could eventually outsource it to its own library.