dionysos-dev/Dionysos.jl #66: Growth bound

blegat, 2021-01-15:
For:
https://github.com/blegat/Dionysos.jl/blob/4bb56a8aa30b34ef76f04c82abdeb74ca62d4fbf/src/controlsystem.jl#L37
why do we use RK4 and not a matrix exponential since F_growthbound is linear in `r` and `u` does not change ?

cc @guberger and @JulienCalbert

guberger, 2021-01-15:
The exact expression would be: e^{L(u)t}*r + \int_0^t e^{L(u)s} sysnoise ds.
I was not sure how the integral could be computed in a numerically stable way.
If L(u) is invertible, the integral is (L(u))^{-1}*(e^{L(u)t} - 1), but it is ill-conditioned, what happens?

blegat, 2021-01-15:
A partial answer is here: https://math.stackexchange.com/questions/658276/integral-of-matrix-exponential but we still need to integrate the exponential of a strictly upper-triangular matrix.

guberger, 2021-01-15:
Oh, not bad JuliaReach! I suppose it comes from the linearization of the system `\dot{x} = Ax + b` into `\dot{x} = Ax + y` and `\dot{y} = 0`, but then I don't get why they have a system of dimension 3n, and not just 2n...

guberger, 2021-01-15:
So in our case, we could have the system `\dot{r} = L(u)r + s` and `\dot{s} = 0` with initial condition `r=r_0` and `s = sysnoise` and thus use the matrix exponential ... (this is what you had in mind?)

blegat, 2021-01-15:
You construct a `3n x 3n` matrix but you only take as subblock of size `n x n`.

guberger, 2021-01-15:
Sure, but a `2n x 2n` would be enough

blegat, 2021-01-15:
You have `e^{L(u)t}*r + \int_0^t e^{L(u)s}*sysnoise ds = e^{L(u)t}*r + Φ₁(u) * sysnoise` where `Φ₁(u)` is the `n x n` matrix given by JuliaReach.

guberger, 2021-01-15:
Or more easily, we compute `[1 0]*exp([A I; 0 0]*t)*[r, sysnoise]` (this is essentially the same idea as in JuliaReach for computing `\phi_1`when A is not invertible)

guberger, 2021-01-24:
I think that in the case of the term `b` being fixed, we can even have a matrix with dimension `(n+1) x (n+1)`, as follows: the solution `x(t)` of the system `\dot{x} = Ax + b` with `x(0)=x_0` would be given by `[1 0]*exp([A b; 0 0]*t)*[x_0, 1]`.

closed 2024-06-27 by guberger