dionysos-dev/Dionysos.jl #171: Update to LazySets v2

blegat, 2022-08-08:
Now `Union` and `UnionSetArray` subtype `LazySet` so this might simplify Dionysos with the issue we had with @adrienbanse

adrienbanse, 2022-09-07:
Seems like non-convex sets are not supported yet although JuliaReach is actively working on it, see:
- https://github.com/JuliaReach/LazySets.jl/issues/3047  
- https://github.com/JuliaReach/LazySets.jl/issues/1895

I propose to keep our implementation for now and extend it if needed

blegat, 2024-10-07:
They now seem to support nonconvex sets so we should reconsider

schillic, 2024-10-20:
On a separate note, LazySets now has submodules for all basic set types. I did not check what exactly you use, but for instance if all you need is the `Ellipsoid`, you could load `LazySets.EllipsoidModule` instead. That could reduce the name clashes (but most public API functions are still exported by each submodule, so it may not help).