dionysos-dev/Dionysos.jl #204: Spring cleaning 🧹

adrienbanse, 2023-05-10:
We identified with @JulienCalbert different things we could clean/implement in the codebase

General
- [ ] Re-write all plotting functions as `Plots.plot` implementing `Plots.jl`, for example 
https://github.com/dionysos-dev/Dionysos.jl/blob/6753f289e377ea9ae3db0398597a9c7c1f874b44/src/utils/ellipsoid.jl#L66

`src/problem`
- [ ] Write module file and implementation file
- [ ] Implement abstract type for problems
- [ ] Check if all problems implement `OptimalControlProblem`
- [ ] In `src/problem/optim`, change to 1 optimizer - 1 file

`src/optim` 
- [ ] Move `src/problem/optim` to `src/optim`
- [ ] Change to 1 optimizer - 1 file (for example SCOTTS should be only one file)
- [ ] Add one folder `src/optim/abstraction` for all abtraction-based methods 
- [ ] Add "NotImplementedError Julia equivalent" if a solver is not implemented for a specific problem

`src/relation`
- [ ] Remove this folder

`src/mapping`
- [ ] Clarify difference between mappings in `src/mapping` and `xpos2int` etc. in symbolic models (@egidioln could you help us with that? Is it really necessary that mapping is a distinct folder in the codebase?)

`src/search`
- [ ] Move this folder to `src/utils/search`

`src/control`
- [ ] Clarify/unify this folder, and point out refactoring tasks for next refactoring (@JulienCalbert, probably a lot of things to do here, we should open another issue for this one)

`src/utils`
- [ ] Check if Dijkstra algorithm in `src/utils/search` is generic enough

adrienbanse, 2023-05-10:
@blegat Does this make sense to you? Don't hesitate to comment

blegat, 2023-05-10:
Yes, that cleaning would make this much clearer!

JulienCalbert, 2023-05-10:
For the question to @egidioln 
Let S1 and S2 be respectively the original system and the abstraction.
We have the three following levels of domains:
domain of S1 (1) <-> discretization of S1 (2) <-> domain of S2 (lists symbols) (3)

There are several way to implement :
- domain S1: periodic or not, rectangular, with obstacles, ...
- discretization of domain  S1: partition vs covering, full vs lazy, ellipsoidal vs polyhedral
- domain of S2: list of symbols
 
For me, the Mapping module should implement the link between (1)<->(2).
While the link between (2)<->(3) should be implemented directly in the symbolicModel (i.e. the structure of the abstraction) since this link depends of the particlar kind of abstraction: hypergraph, simple graph, Markov chain, ...
And I think, it is what you in mind with the code you added in Mapping ?
It makes sense to create a folder for mapping at the same level as the others ?

egidioln, 2023-05-12:
Hello @JulienCalbert and @adrienbanse ! Nice work writing this list of tasks, this cleaning totally makes sense...

> And I think, it is what you in mind with the code you added in Mapping ? 
Ok if you think that adding this level of granularity is too much, but what I was proposing is to have a base class `Mapping` from which many types inheriting from `Mapping` could be created if needed (e.g., `MappingDiscrete2Discrete`, `MappingContinuous2Ellipsoid`, `MappingGraphNode2GraphNode`, ...). This would be a general tool to easily navigate between states, doing like 
```julia
x_closest = MappingDiscrete2Discrete.map(x)
...
ellipsoid_u = MappingContinuous2Ellipsoid.map(u)
...
state_higher_abs = MappingGraphNode2GraphNode.map(state_lower_abs)
```

It can also be use to check/log what we are dealing with:

```julia
if MappingDiscrete2Discrete.Domain
figure_title = "State at " + GraphNode2GraphNode.Origin.Name + "and corresponding ellipsoids in  "" + GraphNode2GraphNode.Origin.Name
...
print("Quantized state x_q:" + MappingC2D(x[k])
```
> It makes sense to create a folder for mapping at the same level as the others ?

Maybe not, maybe `Mapping` could be defined somewhere else (and doesn't have to have its own file\folder)... But as more types may be defined and integrated into Dionysos, it would be good to have them in one place, I think :P

closed 2023-06-13 by adrienbanse