dionysos-dev/Dionysos.jl #211: RigidBodyDynamics.jl- 6. Symbolics using SymPy

JulienCalbert, 2023-05-19:
When I try to execute the following code that Thiago had written (and which is on the L2C drive) about driving a double pendulum with a symbolic representation:

`using Pkg # hide
Pkg.activate(@__DIR__) # hide
Pkg.instantiate() # hide
using RigidBodyDynamics
using StaticArrays
using SymPy

# ## Create symbolic parameters
# * Masses: $m_1, m_2$
# * Mass moments of inertia (about center of mass): $I_1, I_2$
# * Link lengths: $l_1, l_2$
# * Center of mass locations (w.r.t. preceding joint axis): $c_1, c_2$
# * Gravitational acceleration: $g$

inertias = @vars m_1 m_2 I_1 I_2 positive = true
lengths = @vars l_1 l_2 c_1 c_2 real = true
gravitational_acceleration = @vars g real = true
params = [inertias..., lengths..., gravitational_acceleration...]
transpose(params)

# ## Create double pendulum `Mechanism`
# A `Mechanism` contains the joint layout and inertia parameters, but no state information.

T = Sym # the 'scalar type' of the Mechanism we'll construct
axis = SVector(zero(T), one(T), zero(T)) # axis of rotation for each of the joints
double_pendulum = Mechanism(RigidBody{T}("world"); gravity = SVector(zero(T), zero(T), g))
world = root_body(double_pendulum) # the fixed 'world' rigid body

# Attach the first (upper) link to the world via a revolute joint named 'shoulder'
inertia1 = SpatialInertia(CartesianFrame3D("upper_link"),
    moment=I_1 * axis * transpose(axis),
    com=SVector(zero(T), zero(T), c_1),
    mass=m_1)
body1 = RigidBody(inertia1)
joint1 = Joint("shoulder", Revolute(axis))
joint1_to_world = one(Transform3D{T}, frame_before(joint1), default_frame(world));
attach!(double_pendulum, world, body1, joint1,
    joint_pose = joint1_to_world);

# Attach the second (lower) link to the world via a revolute joint named 'elbow'
inertia2 = SpatialInertia(CartesianFrame3D("lower_link"),
    moment=I_2 * axis * transpose(axis),
    com=SVector(zero(T), zero(T), c_2),
    mass=m_2)
body2 = RigidBody(inertia2)
joint2 = Joint("elbow", Revolute(axis))
joint2_to_body1 = Transform3D(
    frame_before(joint2), default_frame(body1), SVector(zero(T), zero(T), l_1))
attach!(double_pendulum, body1, body2, joint2,
    joint_pose = joint2_to_body1)

# ## Create `MechanismState` associated with the double pendulum `Mechanism`
# A `MechanismState` stores all state-dependent information associated with a `Mechanism`.

x = MechanismState(double_pendulum);

# Set the joint configuration vector of the MechanismState to a new vector of symbolic variables
q = configuration(x)
for i in eachindex(q)
    q[i] = symbols("q_$i", real = true)
end

# Set the joint velocity vector of the MechanismState to a new vector of symbolic variables
v = velocity(x)
for i in eachindex(v)
    v[i] = symbols("v_$i", real = true)
end

# ## Compute dynamical quantities in symbolic form

# Mass matrix
simplify.(mass_matrix(x))

# Kinetic energy
simplify(kinetic_energy(x))

# Potential energy
simplify(gravitational_potential_energy(x))`

I get the following error with julia-1.6, while it works with julia-1.5:

ERROR: LoadError: MethodError: no method matching decompose(::Num)
Closest candidates are:
  decompose(::Integer) at float.jl:555
  decompose(::Float16) at float.jl:557
  decompose(::Float32) at float.jl:568
  ...
Stacktrace:
  [1] isfinite(x::Num)
    @ Base .\float.jl:453
  [2] isinf(x::Num)
    @ Base .\float.jl:461
  [3] abs(q::Quaternions.Quaternion{Num})
    @ Quaternions C:\Users\jcalbert\.julia\packages\Quaternions\au6LK\src\Quaternion.jl:143
  [4] sign(x::Quaternions.Quaternion{Num})
    @ Base .\number.jl:135
  [5] QuatRotation
    @ C:\Users\jcalbert\.julia\packages\Rotations\VYTDG\src\unitquaternion.jl:22 [inlined]
  [6] QuatRotation
    @ C:\Users\jcalbert\.julia\packages\Rotations\VYTDG\src\unitquaternion.jl:46 [inlined]
  [7] Rotations.QuatRotation(w::Num, x::Num, y::Num, z::Num, normalize::Bool)
    @ Rotations C:\Users\jcalbert\.julia\packages\Rotations\VYTDG\src\unitquaternion.jl:50
  [8] Rotations.QuatRotation(w::Num, x::Num, y::Num, z::Num)
    @ Rotations C:\Users\jcalbert\.julia\packages\Rotations\VYTDG\src\unitquaternion.jl:49
  [9] rotation_between(from::SVector{3, Num}, to::SVector{3, Num})
    @ Rotations C:\Users\jcalbert\.julia\packages\Rotations\VYTDG\src\unitquaternion.jl:328
 [10] Revolute{Num}(axis::SVector{3, Num})
    @ RigidBodyDynamics C:\Users\jcalbert\.julia\packages\RigidBodyDynamics\8B04X\src\joint_types\revolute.jl:16
 [11] Revolute(axis::SVector{3, Num})
    @ RigidBodyDynamics C:\Users\jcalbert\.julia\packages\RigidBodyDynamics\8B04X\src\joint_types\revolute.jl:26
 [12] top-level scope
    @ C:\Users\jcalbert\Downloads\Thiago-double_pendulum\Symbolic_model.jl:33
 [13] include(fname::String)
    @ Base.MainInclude .\client.jl:444
 [14] top-level scope
    @ REPL[6]:1
in expression starting at C:\Users\jcalbert\Downloads\Thiago-double_pendulum\Symbolic_model.jl:33 

Similarly, the following code in the example does not work in Julia-1.6 either.
[https://github.com/JuliaRobotics/RigidBodyDynamics.jl/tree/master/examples/6.%20Symbolics%20using%20SymPy](url)

blegat, 2023-05-22:
It seems that https://github.com/JuliaGeometry/Quaternions.jl/pull/122 broke our example.
Maybe `isinf` could be defined for Symbolics numbers ?

blegat, 2023-05-22:
A similar error was thrown here: https://github.com/JuliaSymbolics/Symbolics.jl/issues/91 but it was fixed in https://github.com/korsbo/Latexify.jl/pull/145 without defining `isinf` in Symbolics.jl

blegat, 2023-09-05:
This issue is relevant: https://github.com/JuliaSymbolics/Symbolics.jl/issues/97
We can just use type piracy for the methods that are missing