dionysos-dev/Dionysos.jl #218: Fix plot in docs

JulienCalbert, 2023-05-21:
See https://github.com/dionysos-dev/Dionysos.jl/pull/212, gets the error: https://dionysos-dev.github.io/Dionysos.jl/dev/generated/DC-DC%20converter/#:~:text=qt.qpa.xcb,GKS%20socket%20application

_qt.qpa.xcb: could not connect to display
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: linuxfb, minimal, offscreen, vnc, xcb.

Aborted (core dumped)
connect: Connection refused
GKS: can't connect to GKS socket application_

blegat, 2023-05-22:
This seems related: https://discourse.julialang.org/t/generation-of-documentation-fails-qt-qpa-xcb-could-not-connect-to-display/60988
I can't say more because discourse seems to be down at the moment.

JulienCalbert, 2023-06-08:
Hi Benoît and thanks for the link.
I implemented the solution that I found there 
https://github.com/simonp0420/PSSFSS.jl/blob/9d9a69b0e86f51ac261b6f55ce26d6305127a3f9/.github/workflows/CI.yml#L64-L67
By just adding the line in the CI.yml file
GKSwstype: "100" # https://discourse.julialang.org/t/generation-of-documentation-fails-qt-qpa-xcb-could-not-connect-to-display/60988

In local it works, but we get an error on github (like invalid syntax) ... 
Do you have an idea ? Thanks !

JulienCalbert, 2023-06-08:
@blegat it's working, it was just an indentation problem, thanks @adrienbanse !

adrienbanse, 2023-06-08:
New problems: 
1) The plots display after each cell, and do not display when `display(fig)` is called. Maybe `plot!(fig, ...)` instead of `plot!(...)` solves the problem
2) The plots do not look alike the local notebooks

adrienbanse, 2023-06-09:
https://github.com/dionysos-dev/Dionysos.jl/pull/228 solves most of the issue
@JulienCalbert Just remains to fix the Ellipsoids plot, I'm on it

closed 2023-06-09 by adrienbanse