dionysos-dev/Dionysos.jl #221: Deprecated `examples` folder

adrienbanse, 2023-05-22:
We should update the `examples` folder (for example with the new `Optim` module, see https://github.com/dionysos-dev/Dionysos.jl/pull/220), and write tests corresponding to these examples in order to keep it updated.

adrienbanse, 2023-06-06:
In the end, the idea would be to **remove this folder** and translate a maximum amount of this either in example in the doc, or in tests

adrienbanse, 2023-09-05:
cc @blegat

blegat, 2023-09-06:
As discussed with Julien, most of it is used as plotting utilities for slides and presentations so we could rename it `utils` and have each file explain what it does as suggested in https://github.com/dionysos-dev/Dionysos.jl/pull/287
If the file can be tested then we could also add it in the test folder

closed 2025-04-24 by JulienCalbert