dionysos-dev/Dionysos.jl #226: Implement text in plots with recipes

adrienbanse, 2023-06-08:
I don't know where this is used, but it could be incorporated in a recipe as well with an optional argument
Interestingly recipees are not only used to define a default plot for a structure, but also to define other type of plotting functions, but where one can still use all the classical plotting arguments   

See "Plot Recipes" and not "Type Recipes" here https://docs.juliaplots.org/latest/recipes/

_Originally posted by @adrienbanse in https://github.com/dionysos-dev/Dionysos.jl/pull/225#discussion_r1222577533_

blegat, 2023-09-08:
Can't we replace the Plots.jl dependency by RecipesBase.jl now ?

adrienbanse, 2023-09-08:
@blegat We still have some plotting functions in `src/control/controller.jl`

adrienbanse, 2023-09-08:
Needs to be cleaned anyway, see https://github.com/dionysos-dev/Dionysos.jl/issues/286