dionysos-dev/Dionysos.jl #233: Bug Gol-Lazar-Belta on Julia 1

adrienbanse, 2023-06-09:
@blegat Seems like the Gol-Lazar-Belta bugs only on Julia 1 (not 1.6). Locally I have no problem.
If I re-run the error can disappear.. Any idea? Seems related to Pavito
\
\
\
\
\
Depth: 11: Test Failed at /home/runner/work/Dionysos.jl/Dionysos.jl/test/examples/test_gol_lazar_belta.jl:31
  Expression: MOI.get(optimizer, MOI.TerminationStatus()) in [MOI.OPTIMAL, MOI.LOCALLY_SOLVED]
   Evaluated: MathOptInterface.ALMOST_LOCALLY_SOLVED in MathOptInterface.TerminationStatusCode[MathOptInterface.OPTIMAL, MathOptInterface.LOCALLY_SOLVED]

Stacktrace:
 [1] macro expansion
   @ /opt/hostedtoolcache/julia/1.9.1/x64/share/julia/stdlib/v1.9/Test/src/Test.jl:478 [inlined]
 [2] _test(algo::MathOptInterface.OptimizerWithAttributes, N::Int64, q0::Int64, x0::Vector{Float64}, x_expected::Vector{Vector{Float64}}, u_expected::Vector{Vector{Float64}}, obj_expected::Float64, zero_cost::Bool, mi::Bool; kws::Base.Pairs{Symbol, Union{}, Tuple{}, NamedTuple{(), Tuple{}}})
   @ Main.TestMain ~/work/Dionysos.jl/Dionysos.jl/test/examples/test_gol_lazar_belta.jl:31
Depth: 11: Error During Test at /home/runner/work/Dionysos.jl/Dionysos.jl/test/examples/test_gol_lazar_belta.jl:151
  Got exception outside of a @test
  Result index of attribute MathOptInterface.VariablePrimal(1) out of bounds. There are currently 0 solution(s) in the model.
  Stacktrace:
    [1] check_result_index_bounds
      @ ~/.julia/packages/MathOptInterface/BlCD1/src/attributes.jl:207 [inlined]
    [2] get
      @ ~/.julia/packages/Pavito/bhsCM/src/MOI_wrapper.jl:312 [inlined]
    [3] get(b::MathOptInterface.Bridges.LazyBridgeOptimizer{Pavito.Optimizer}, attr::MathOptInterface.VariablePrimal, index::MathOptInterface.VariableIndex)
      @ MathOptInterface.Bridges ~/.julia/packages/MathOptInterface/BlCD1/src/Bridges/bridge_optimizer.jl:1195
    [4] get(model::MathOptInterface.Utilities.CachingOptimizer{MathOptInterface.AbstractOptimizer, MathOptInterface.Utilities.UniversalFallback{MathOptInterface.Utilities.Model{Float64}}}, attr::MathOptInterface.VariablePrimal, index::MathOptInterface.VariableIndex)
      @ MathOptInterface.Utilities ~/.julia/packages/MathOptInterface/BlCD1/src/Utilities/cachingoptimizer.jl:911
    [5] _value
      @ ~/work/Dionysos.jl/Dionysos.jl/src/optim/bemporad_morari.jl:110 [inlined]
    [6] _broadcast_getindex_evalf
      @ ./broadcast.jl:683 [inlined]
    [7] _broadcast_getindex
      @ ./broadcast.jl:656 [inlined]
    [8] getindex
      @ ./broadcast.jl:610 [inlined]
    [9] copy
      @ ./broadcast.jl:912 [inlined]
   [10] materialize(bc::Base.Broadcast.Broadcasted{Base.Broadcast.DefaultArrayStyle{2}, Nothing, typeof(Dionysos.Optim.BemporadMorari._value), Tuple{Base.RefValue{MathOptInterface.Utilities.CachingOptimizer{MathOptInterface.AbstractOptimizer, MathOptInterface.Utilities.UniversalFallback{MathOptInterface.Utilities.Model{Float64}}}}, Matrix{MathOptInterface.VariableIndex}}})
      @ Base.Broadcast ./broadcast.jl:873
   [11] get(optimizer::Dionysos.Optim.BemporadMorari.Optimizer{Float64}, #unused#::Dionysos.Control.ContinuousTrajectoryAttribute)
      @ Dionysos.Optim.BemporadMorari ~/work/Dionysos.jl/Dionysos.jl/src/optim/bemporad_morari.jl:349
   [12] _test(algo::MathOptInterface.OptimizerWithAttributes, N::Int64, q0::Int64, x0::Vector{Float64}, x_expected::Vector{Vector{Float64}}, u_expected::Vector{Vector{Float64}}, obj_expected::Float64, zero_cost::Bool, mi::Bool; kws::Base.Pairs{Symbol, Union{}, Tuple{}, NamedTuple{(), Tuple{}}})
      @ Main.TestMain ~/work/Dionysos.jl/Dionysos.jl/test/examples/test_gol_lazar_belta.jl:32
   [13] _test
      @ ~/work/Dionysos.jl/Dionysos.jl/test/examples/test_gol_lazar_belta.jl:21 [inlined]
   [14] #_test11#4
      @ ~/work/Dionysos.jl/Dionysos.jl/test/examples/test_gol_lazar_belta.jl:119 [inlined]
   [15] _test11
      @ ~/work/Dionysos.jl/Dionysos.jl/test/examples/test_gol_lazar_belta.jl:118 [inlined]
   [16] macro expansion
      @ ~/work/Dionysos.jl/Dionysos.jl/test/examples/test_gol_lazar_belta.jl:152 [inlined]
   [17] macro expansion
      @ /opt/hostedtoolcache/julia/1.9.1/x64/share/julia/stdlib/v1.9/Test/src/Test.jl:1498 [inlined]
   [18] macro expansion
      @ ~/work/Dionysos.jl/Dionysos.jl/test/examples/test_gol_lazar_belta.jl:152 [inlined]
   [19] macro expansion
      @ /opt/hostedtoolcache/julia/1.9.1/x64/share/julia/stdlib/v1.9/Test/src/Test.jl:1586 [inlined]
   [20] (::Main.TestMain.var"#tests#5")(qp_solver::MathOptInterface.OptimizerWithAttributes, miqp_solver::MathOptInterface.OptimizerWithAttributes)
      @ Main.TestMain ~/work/Dionysos.jl/Dionysos.jl/test/examples/test_gol_lazar_belta.jl:127
   [21] macro expansion
      @ ~/work/Dionysos.jl/Dionysos.jl/test/examples/test_gol_lazar_belta.jl:184 [inlined]
   [22] macro expansion
      @ /opt/hostedtoolcache/julia/1.9.1/x64/share/julia/stdlib/v1.9/Test/src/Test.jl:1498 [inlined]
   [23] top-level scope
      @ ~/work/Dionysos.jl/Dionysos.jl/test/examples/test_gol_lazar_belta.jl:125
   [24] include(fname::String)
      @ Base.MainInclude ./client.jl:478
   [25] top-level scope
      @ ~/work/Dionysos.jl/Dionysos.jl/test/runtests.jl:33
   [26] include(fname::String)
      @ Base.MainInclude ./client.jl:478
   [27] top-level scope
      @ none:6
   [28] eval
      @ ./boot.jl:370 [inlined]
   [29] exec_options(opts::Base.JLOptions)
      @ Base ./client.jl:280
   [30] _start()
      @ Base ./client.jl:522

adrienbanse, 2023-06-09:
Maybe related to the switch to HiGHS https://github.com/dionysos-dev/Dionysos.jl/issues/209

adrienbanse, 2023-06-09:
Thanks for your reply.
Any idea where this is coming from? Is there something random in the solver?

blegat, 2023-06-09:
Not in Pavito itself but maybe in one of the inner solvers.

closed 2023-06-09 by adrienbanse