dionysos-dev/Dionysos.jl #242: Mosek

JulienCalbert, 2023-06-15:
to use Mosek in Github Actions for docs and tests, we use http://solve.mosek.com (see https://github.com/MOSEK/Mosek.jl/issues/206) but it's very slow.
It remains to be seen whether we can find a solver to replace Mosek that doesn't require a license.

adrienbanse, 2023-06-15:
FYI here is a list of solvers, I don't know which ones may suit and have a JuMP wrapper

https://yalmip.github.io/allsolvers/

adrienbanse, 2023-06-15:
Also, @JulienCalbert, Mosek and MosekTools should'nt be in Dionysos env, but only in the test environnement in [extras] and [targets]
And I don't think anyone reviewed https://github.com/dionysos-dev/Dionysos.jl/pull/240 right ?

JulienCalbert, 2023-06-15:
oups yes indeed

adrienbanse, 2023-06-15:
Here is a list to test

- [x] ~~CDCS~~ (does not support log det)
- [x] ~~Clarabel~~ (does not support mixed)
- [x] ~~COPT~~ (under license)
- [x] ~~COSMO~~ (does not support log det)
- [x] ~~CSDP~~ (does not support log det)
- [x] **Hypatia ~ 68s on my machine**
- [x] MOSEK ~ 78s on my machine
- [x] ~~Pajarito~~ (does not support log det)
- [x] ~~ProxSDP~~ (does not support log det)
- [x] SCS ~ 184s on my machine
- [x] ~~SDPA~~ (MATLAB license)
- [x]  ~~SDPNAL~~ (MATLAB license)
- [x] ~~SeDuMi~~ (MATLAB license)

Let's go for Hypatia then @JulienCalbert  ?

adrienbanse, 2023-06-15:
@JulienCalbert Seems like Hypatia needs more iteration to find the same solution, so I propose we keep Mosek with the online server for now.. We don't have much choice

JulienCalbert, 2023-06-15:
great job comparing these solvers Adrien ! Thanks a lot!

egidioln, 2023-06-15:
@adrienbanse great job! I did a similar benchmark a few years ago and SDPA was the best one that didn't require a license. Does it need a Matlab license now?

Also, for those that don't support log det, a frank wolf - like algorithm can be used, which solves a min c'x problem at each iteration and converges quite fast

adrienbanse, 2023-06-16:
My bad @egidioln, it's a typo! It just doesn't support LogDetConeTriangle.
You say there exists an alternative formulation for which we could use all the solvers that do not support log det?

adrienbanse, 2023-06-16:
I keep having this error when I try to do the following
```
using SDPA
sdp_opt = MOI.instantiate(
    MOI.OptimizerWithAttributes(SDPA.Optimizer, MOI.Silent() => true), 
    with_bridge_type = Float64
)
```

ERROR: The provided `optimizer_constructor` is invalid. It must be callable with zero arguments. For example, "Ipopt.Optimizer" or "() -> ECOS.Optimizer()". It should not be an instantiated optimizer like "Ipopt.Optimizer()" or "ECOS.Optimizer()". (Note the difference in parentheses!)

Any idea why?

adrienbanse, 2023-06-19:
After a fair amount of considered alternatives, let's keep the online solver of Mosek for the tests in Github Actions

closed 2023-06-19 by adrienbanse