dionysos-dev/Dionysos.jl #282: Cleaning before release

adrienbanse, 2023-09-05:
- Define what files/functions should be cleaned/removed @adrienbanse 
- Write documentation for core Dionysos @SMattenet @vdebauche 
- Write Dionysos manual @JulienCalbert

vdebauche, 2023-09-07:
I try to add documentation for each Optimizer in the optim folder (BemporadMorari, BranchAndBound and the abstraction based Optimizer) but when I want to add them in the documentation using "Dionysos.Optim.BranchAndBound.Optimizer", it does not work. Am I doing something wrong? @blegat any idea ?

adrienbanse, 2023-09-07:
https://github.com/dionysos-dev/Dionysos.jl/pull/291 already structures the documentation, and defines what should be further documented

JulienCalbert, 2023-09-08:
https://github.com/dionysos-dev/Dionysos.jl/pull/294 first version of the manual.

adrienbanse, 2023-09-08:
https://github.com/dionysos-dev/Dionysos.jl/pull/292 adds more documentation

adrienbanse, 2023-10-23:
🥳