dionysos-dev/Dionysos.jl #302: RangeEnclosures

blegat, 2023-09-13:
We could use [RangeEnclosures](https://github.com/JuliaReach/RangeEnclosures.jl) for computing the abstraction.
It has an interface with a list of solvers so if our abstraction based solvers are parametrized by a `AbstractEnclosureAlgorithm` then they automatically support any alogrithm implementing the `RangeEnclosures` API !
It's still only for scalar output but in [their paper](https://github.com/JuliaCon/proceedings-review/issues/122), they say vector output is planned, let's keep an eye on it !
If what we still want to use what we are currently doing by default, we could implement it as a solver and have it as default solver.

cc @schillic @mforets @lucaferranti

schillic, 2024-03-23:
Just a heads up that this already works if you can deal with tuple outputs:

```julia
# 1D input, 2D output

julia> f(x) = (x, 2x);

julia> X = interval(1..2)
[1.0, 2.0]

julia> f(X)
(Interval{Float64}(1.0, 2.0), Interval{Float64}(2.0, 4.0))

# 2D input, 2D output

julia> g((x, y)) = (x, 2y);

julia> Y = IntervalBox(1..2, 3..4)
[1.0, 2.0] × [3.0, 4.0]

julia> g(Y)
(Interval{Float64}(1.0, 2.0), Interval{Float64}(6.0, 8.0))
```

schillic, 2024-03-24:
I also just added a PR for better support of vector outputs, at least for some of the solvers, in https://github.com/JuliaReach/RangeEnclosures.jl/pull/175. But it may require some discussions before that gets in.