dionysos-dev/Dionysos.jl #345: Add the computation of a quadratic Lyapunov function

JulienCalbert, 2024-04-15:
To plot the bisimilar abstraction for switched systems (DCDC converter), and compute the abstraction parameters, we need to compute a common quadratic lyapunov function.

blegat, 2024-04-15:
That's the overarching goal of https://github.com/blegat/SwitchOnSafety.jl but it the moment it only supports a limited set of systems so you can add it in Dionysos and we can see later if we can refactor

adrienbanse, 2024-04-15:
Isn't it simply a SDP solver with bissection?

blegat, 2024-04-15:
Yes but it's quite tricky to get it right, see
https://github.com/blegat/SwitchOnSafety.jl/blob/master/src/sos.jl
First you want to do the bisection in the log scale. Second (especially with degree higher than 2), you want to divide the matrices (describing the dynamical system) by γ and not multiply the LMI by γ^d otherwise the scaling is bad.
Last but not least, there will be a blind spot around the critical γ where the SDP goes from infeasible to feasible and your bisection should be robust against that.
The struggle I had for computing the JSR numerically with SDP solvers was one of the motivation of the better status reporting in MOI compared to MathProgBase ^^